fruit-bat/pico-zxspectrum #141: ILI9143 LCD

salvadooor, 2024-06-16:
Hi! I want to use an Ili9143 lcd display. The link (https://github.com/fruit-bat/pico-zxspectrum/blob/main/docs/ZxSpectrumPicomputerZxLcd.md
) does not contain the file (ZxSpectrumPicomputerZxILI9341Lcd.uf2).
Can you help me? 

Thank you!

fruit-bat, 2024-06-16:
I've updated https://github.com/fruit-bat/pico-zxspectrum/blob/main/docs/ZxSpectrumPicomputerZxLcd.md to point at the correct firmware (hopefully). There are a couple of options for the board for different VGA modes. Let me know if it works :-)

salvadooor, 2024-06-16:
Thank you very much for your quick help! I copied the ZxSpectrumPicomputerZxILI9341Lcd_640x480x60Hz.uf2 firmware and it works! :) 
Instead of external keyboard and full matrix keyboard, I want to use only mini panel joystick ( 5 buttons: 4 direction + 1 fire) 
Where can I find the keyboard matrix for this firmware to see if the fire button is which column+row? (Do I only need to connect the two pins together to connect or do I need to connect to gnd/vcc?)
Thank you very much!

fruit-bat, 2024-06-17:
> Thank you very much for your quick help! I copied the ZxSpectrumPicomputerZxILI9341Lcd_640x480x60Hz.uf2 firmware and it works! :) Instead of external keyboard and full matrix keyboard, I want to use [...]

I've added a schematic, hopefully that makes it clear :-)

salvadooor, 2024-06-24:
Thank you! Based on the keyboard matrix I managed to hook up a 4-way pushbutton-joystick + fire and menu button. I can go up and down in the menu but I can't understand the tape loading. 

1, in the tape loader menu there is only ?????????.??? in the tap files place. i can go up and down but between them i can't select and load (and ???????.??? i don't know which game :) ). (I tried 512Mb SD card, 4Gb SD card, Fat and exFat)

2, I tried plugging in a USB keyboard to see if that solves the selection problem. But I couldn't get it to work at all :( 
Usb keyboard power gnd and Vsys 5V.
D+ - DP0 , D- - DP1. I tried directly connecting D- to D+, then tried using series resistor, then pulling up resistor to positive. The keypad led lights up but nothing works with either connection unfortunately. 

What should I do? 

Thanks for your help!

fruit-bat, 2024-06-24:
The question marks mean it needs to refresh the folder. There is on option to do this (the one ggat says REF, I think that's number 5)... but you will need a working keyboard.  I've always used an OTG adapter so I'm not sure if there is a 'trick' to getting host mode to work... I'll have a Google!

fruit-bat, 2024-06-24:
Looks like you need to ground the sense at the pico end...

https://onlinediscount.sales2024tore.com/menu?name=micro+usb+otg+pinout#dk=micro%60usb&line=527

salvadooor, 2024-06-25:
Thanks for all your help ! I'm trying to search for info in google too, but of course I can't find a solution for specific hardware and software.

I must be doing something wrong ! 
I am trying to describe the wiring and the current operation.

1, According to the matrix keyboard I have connected a button to refresh (5). If it refreshes then the ????????.???? , the menu will be empty. 

2, I cut the keyboard usb cable, there are only 4 wires. There is no external ground (sense) so I can't connect it. 

My keyboard wiring:
USB red - Pico PI VBus
USB black - Pico PI GND
USB green - Pico PI Pin2
USB white - Pico PI Pin1
Pico PI Pin1 and Pin2 with 10kohm resistor connected 3.3V 

.... Power from Pico PI usb C

The keyboard 3led lights up on startup. The keyboard does not work at all.

Thank you!

fruit-bat, 2024-06-25:
Ok, first let me just check the build you are using still works with usb... I (sadly) have real work to do this evening so may be tomorrow or so.

fruit-bat, 2024-06-26:
Ok, so I know this isn't a great help but I have the firmware and can confirm that I can connect a USB keyboard and navigate the menus with it. 

I don' t think you can be far off; maybe try a different keyboard. Try connecting through a hub... some devices don't like being directly connected.

salvadooor, 2024-06-27:
Can you share your USB connection? 

Are there any series resistors on the D- and D+ pins? Pull-up resistor on any of the pins? 

USB -, +, D-, D+ ... PICO PI Pin? Pin? etc  How are they connected? 

Thank you!

fruit-bat, 2024-06-27:
The circuit diagram for Peter's board is here:
https://github.com/fruit-bat/pico-zxspectrum/blob/main/docs/ZxSpectrumPicomputerZxLcd.md

I generally work on a breadboard and use an otg connector to my usb keyboard... so no extra wiring... but otg connectors have some internal connections... https://en.m.wikipedia.org/wiki/USB_On-The-Go#:~:text=An%20OTG%20cable%20has%20a,micro%2DB%20plug%20is%20floating.

[image]

salvadooor, 2024-06-27:
1, Yes, I connected it according to the wiring diagram first. Directly D- is the Pi Pin1 and D+ is Pin2. But on the schematic of Peter there is no resistor for USB. Is that ok?

2, Is it OK that Pin1 and Pin2 are also connected to the VGA in the wiring diagram?

3, Or did you connect the USB keyboard to the Micro USB port of the PICO PI and use it that way? 
( that's not how it's marked on the wiring diagram if I saw it correctly)

Thank you!

fruit-bat, 2024-06-27:
I've tried both the USB port of the PICO PI  (with the OTG connector) and the USB A port on Peter's board. Regardless of what the circuit diagram says, from looking at the board the USB A connector is wired to the little test pads on the other side of the PICO PI's micro USB B connector; I think it's just the D+ and D- connected like this.

salvadooor, 2024-06-28:
1, Now the keyboard works! :)
Usb keyboard is soldered to the PICO PI microUsb islands.

2, SD card - Tape list
Checked the spi connection of the Sd card, good. Card is FAT32. Tried several cards Fat and FAT32. Reads, but unfortunately the tape list is still ???????.???. If I refresh the list with "5" button the whole list disappears.  I tried restarting many times but no change.

salvadooor, 2024-06-28:
1, At first the list was just as bad and didn't update, but after 2 more SD card tries it did! 
The list is good! :) Thank you!

2, I tried loading several games with TAP and TZX files. It starts to load but stops e.g. "tape loading error, 10:1" or other games with other errors. 
I tried zxspectrum 48k game and 128k game.

3, Unfortunately when I select 48k in the menu I can't select the tape loader and load from there after rebooting. Only the basic spectrum command line will accept. What should I do if I want to load a 48k game?

Translated with www.DeepL.com/Translator (free version)

fruit-bat, 2024-06-29:
Sorry, not sure what is going wrong there. The tape loading is  not perfect but most of the tap files I try load successfully. 
In 48k mode, you need to type the following on the keyboard to load from tape:

J Alt-Gr+P Alt-Gr+P ENTER

Where Alt-Gr is the right hand Alt key on the USB keyboard.

Are you still using ZxSpectrumPicomputerZxILI9341Lcd_640x480x60Hz.uf2 ? ... I will check I have not done anything stupid in the config for it.

salvadooor, 2024-06-29:
Yesterday I managed to use the 48k mode loading. Unfortunately it didn't load completely once, but it got a lot done. 

Yes, I am using this rom (ZxSpectrumPicomputerZxILI9341Lcd_640x480x60Hz.uf2), the other roms didn't work for me. The 50hz reboots quickly, and the others have inverse display and color.

fruit-bat, 2024-07-02:
I wonder if this is because the rp2040 is overclocked? All mine seem to work fine, but it is possible. Perhaps try a different power supply. Maybe add  a small decoupling capacitor.