fruit-bat/pico-zxspectrum #142: 4BK - 4 Buttons Kempston Joysick

javavi, 2024-06-20:
Many modern, new games use the 8-bit Kempston joystick.
Spectral Interlude (2015)
The Dark by Oleg Origin (1997, 2016)
Mighty Final Fight (2017)
Aliens: Neoplasma (2019)
Delta's Shadow (2020)
Vade Retro II by J.B. (2017)
The Order of Mazes (2015)
Hell Yeah! (2020)
and so on..
I suggest making the following 4B Kempston joystick layout - |Y|X|A|B|.|D|D|D|D|

fruit-bat, 2024-07-14:
See 
https://github.com/fruit-bat/pico-zxspectrum/pull/152/files
https://github.com/fruit-bat/pico-zxspectrum/tree/feature/joyhat

Let me know if it works for you

javavi, 2024-07-14:
> Let me know if it works for you

Yes, this Hat works on both my Joysticks :))
Thank you very much

javavi, 2024-08-24:
In the game Neo Plasma the grenade throw button (6th bit Kempston joystick) stopped working, instead it exits to the emulator menu.
To exit to the emulator menu it would be better to use a combination of holding down two buttons at the same time (as is done in many emulators)
Please do so.

fruit-bat, 2024-08-24:
I've push some changes. Please let me know if they help.

javavi, 2024-08-24:
> I've push some changes. Please let me know if they help.

Yes! That's how it turned out, in my opinion, great!
And the game works and you can exit the menu with the joystick.
Both the PC USB joystick and the Xbox controller work.

closed 2024-08-31 by javavi