fruit-bat/pico-zxspectrum #144: Fast loading

jackic2, 2024-06-23:
Does this software support fast tape loading? I could not find any reference to it on the documentation, thanks for the attention.

fruit-bat, 2024-06-23:
Just, unlimit the cpu speed. Should go quite a bit faster.

javavi, 2024-06-29:
Instant loading of TAP files is possible by intercepting an interrupt to the bootloader inside the Spectrum ROM.
This is implemented in other projects.

javavi, 2024-07-08:
I came across programs that during their loading required stopping the tape to confirm some actions or created a pause themselves, and since the TAP player continued to play the file, further loading was interrupted. It would be good to have a play/pause button on the PC keyboard to control playback by the TAP player?

fruit-bat, 2024-07-08:
> I came across programs that during their loading required stopping the tape to confirm some actions or created a pause themselves, and since the TAP player continued to play the file, further [...]

It is supposed to. See https://github.com/fruit-bat/pico-zxspectrum/blob/main/src/PulseProcTzxBlock.cpp around line 197.

Need to get a listing of the contents of the file using tzxcat (I think). 

https://shredzone.org/docs/tzxtools/

[grr, installing python programs is such a pain now]

fruit-bat, 2024-07-12:
The command to list the contents of the tape file is actually:
```
tzxls.py -v filename.tzx
```
The command to stop the tape drive is a pause of 0ms, as far as I understand. I've tried a couple of examples and the tape motor does indeed switch to off. 

Maybe have a look at the contents of the file causing trouble and see if it actually has a pause 0ms in it?

javavi, 2024-07-15:
Here I asked a friend (@tecnocat) from another project ZX-Spectrum for an example of a procedure for fast automatic loading of TAP files. Perhaps it is possible to make an option for normal and fast loading of programs in TAP format.
[fast_tape_load.zip](https://github.com/user-attachments/files/16233880/fast_tape_load.zip)

fruit-bat, 2024-07-15:
Many thanks, I will have look and see if it can be added.