fruit-bat/pico-zxspectrum #147: Adafruit Feather RP2040 DVI

cgreening, 2024-06-28:
Is there out of the box support for the Adafruit Feather RP2040 DVI?

I'm happy to try and add this if someone can point me in the right direction - I have no idea how cmake works, so I'm slightly lost...

https://learn.adafruit.com/adafruit-feather-rp2040-dvi/pinouts

fruit-bat, 2024-06-29:
Not supported at the moment, and very happy for you to have a go at adding support.

Fork yourself a copy of the project then start by getting to a point you can build an existing target... e.g. 
`
make -j4 ZxSpectrumBreadboardHdmiAudio720x576x50Hz
`

Then have a look in :
https://github.com/fruit-bat/pico-zxspectrum/blob/feature/adafruit-feather-dvi/src/hdmi/CMakeLists.txt

You *may* be able to set everything up from in there, just by following the pattern for the other boards.

I've created a branch for the work:
https://github.com/fruit-bat/pico-zxspectrum/tree/feature/adafruit-feather-dvi

Work on this branch and pull request to my branch (This way I can try stuff out without breaking main).

Cheers,

Phil

cgreening, 2024-06-29:
Great - I have something that semi works. It seems to freeze quite often on boot.

Not sure if it's the pins I've chosen for the SD Card or something else.

Are there any limits on what pins can be used on the RP2040? I'm much more familiar with the ESP32.

fruit-bat, 2024-06-29:
The following diagram is useful:
https://github.com/fruit-bat/pico-zxspectrum/blob/main/docs/Pico-R3-SDK11-Pinout.svg

I don't think the other builds freeze on boot... so there may be something up with your configuration (or something else!!). Can you read from the SD card? Have you substituted in my version of TinyUSB into your pico-sdk?

fruit-bat, 2024-06-29:
Did you do design the ESP32 Spectrum board with the lovely printed keyboard?

cgreening, 2024-06-29:
I'm starting to wonder if it's the feather dvi board. It works find when it's first flashed, but as soon as I hit the reset button or power cycle it. I get a blank screen. And now it's stopped working at all... oops.

Yes, that's me. I was thinking of connecting to an RP2040 over I2C and adding DVI output to the board. But I've now killed my feather...

I think I'll try with one of the boards that is known to be working already... or maybe just start with the breadboard version.

fruit-bat, 2024-06-29:
In theory, the rp2040 has enough pins for the key matrix, hdmi (with audio) and sd card (unless I am rubbish at counting, which is very possible).

cgreening, 2024-06-29:
Main problem would be the capacitive touch pads - I know there's are ways to make touch work with GPIO pins, but also know it definitely works out of the box with the ESP32.

At the moment, I'm just playing with ideas for adding more options to the board - a plugin "graphics card" that provides DVI output (or even something built into the board that does it) would be a nice thing to have.

Waiting on the next board to come back from china so have some time to play with options.

It could end up the ESP32 turns into a keyboard/IO controller for a RP2040.

But, I do need to get the board fully working :)

I'm going to close this for now as I don't have a feather to test with anymore.

cgreening, 2024-07-01:
My board is working again. I'll get a PR up.

closed 2024-07-03 by cgreening