fruit-bat/pico-zxspectrum #158: Pico Pimoroni VGA demo base USB keyboard

sonietje, 2024-07-31:
Hi,
I managed to connect a USB keyboard to the pico but non-letter shift keys are not working. When I press 'shift'+';' I get nothing while I expect ':'. This is the same for all special characters.
But shift + letter shows the upper case. 
I tried with different USB keyboards but no luck. 
Thanks for looking into this.

fruit-bat, 2024-07-31:
The emulator map keys on the usb keyboard to zx spectrum keys (mostly, in a few cases a single usb key is mapped to multiple spectrum key presses). The effect of shift will be as if the spectrum key was shifted. Is this what you are seeing or have I misunderstood the problem?

sonietje, 2024-08-01:
I'm using a UK usb keyboard. So if I want the double quote I press shift single quote but nothing happens. This is the same for all the special character keys.

DPRCZ, 2024-08-01:
Try boot to 128K mode and  selec basic. There should Letter key write letter.

sonietje, 2024-08-02:
I think I get it. The special characters I need such as ':' and '"' are not in the shift position of the fysical key of the USB keyboard but on the keys of the ZX keyboard. So '"' is right alt key + 'P' and the ':' is right alt + 'Z'. I think I can live with this 😁!
Thanks for the clarification.