fruit-bat/pico-zxspectrum #159: Any Usb stick or smartphone connection is available?

upiter20, 2024-08-03:
Good day. I am interested in subj!

fruit-bat, 2024-08-04:
At the moment, only indirectly. The firmware versions that support an audio input can load from the sound of a tape. There are Android apps to do this, like PlayZX. Connection would be made with an audio cable.

I think the TinyUSB library I use includes support for USB storage, but there would still be a reasonable amount of work to get it integrated, so unlikely to happen in the short term... unless someone gets round to it before me!

pauldude12, 2025-07-20:
See these uses TTL data shifters digital signals like the c64 tape deck does it called a adc dac analogue to digital and digital to analogue ic chip