fruit-bat/pico-zxspectrum #161: Help identifying my particular board for update

jackic2, 2024-08-12:
Hi, I bought this prebuilt board from ebay:

https://www.ebay.es/itm/226132204892

I just assumed this would use the "ZxSpectrumPicomputerVgaAukBob" so I flash it to the pico and I just get no video signal, not a black screen, just no video signal (according to my monitos that is), I looked for a similar board on the internet and found this one:

https://www.pcbway.com/project/shareproject/PICOZX_motherboard_for_ZX_Spectrum_original_case_5bbde8be.html

That seems to be identical, I just downloaded the firmware on the pcbway website and it just works, the file is labelled "ZxSpectrumPicomputerVgaAukBob_PCBWay Community.uf2" that seems to be the same as the one on this github page, am I doing something wrong? Did I identify the board correctly?

Any help will be greatly appreciated, thanks.

fruit-bat, 2024-08-13:
Ok, thanks for the detail. I will see if I can fix whatever has gone wrong. There is a new SDK for the RP2350 so maybe something there has upset things...

fruit-bat, 2024-08-17:
@jackic2 hopefully, the VGA builds are up and running again. If you click on the picture of your board on this [page](https://github.com/fruit-bat/pico-zxspectrum) it should link to another page with information about which firmware is suitable for the board. Failing that post a picture of your board and I will have a guess for you :-)