fruit-bat/pico-zxspectrum #163: Could you move your version TinyUSB to 0.16.0 ?

DPRCZ, 2024-08-17:
I like to add support XINPUT device with  Ryzee119 /  tusb_xinput,  but it need newer TinyUSB.

fruit-bat, 2024-08-17:
I'll have a go... don't hold your breath... merging in the latest updates from TinyUSB can be really difficult.

fruit-bat, 2024-08-21:
Thanks for the pull request. I've not merged it as I am using my master to track hathach's master (might be pointless but it's what I've done so far).

I've created a new branch on my fork of TinyUSB: merge/hid_micro_parser_21_08_24

This should contain my changes merged with the latest from hathach (apologies if this has duplicated effort from your pull request). 

It seems to work ok for me. Please let me know how you get on with it. If you think it is ok I will merge the changes to the branch: hid_micro_parser

DPRCZ, 2024-08-21:
TinyUSB: merge/hid_micro_parser_21_08_24 seems to work OK, USB HID gamepad check, XINPUT gamepad check, USB Keyboard check, With USB mouse I've only tested usb mount - because it doesn't work in MENU.

fruit-bat, 2024-08-23:
Thanks for the feedback.  I will check the mouse is working...

fruit-bat, 2024-08-23:
Mouse seems to be working ok. Changes merged back to hid_micro_parser branch.

closed 2024-08-24 by DPRCZ