fruit-bat/pico-zxspectrum #167: After "Switch to scanline VGA" stopped my board working

DPRCZ, 2024-08-21:
I'm using  build "ZxSpectrumPicomputerVgaAukBob_640x480x60Hz" . I've update git "https://github.com/fruit-bat/pico-extras".

I'm still use SDK 1.0 could be issue ??? 

Afer build firmware and download to RP2040 VGA doesn't start work - monitor is black  and  buzzer make a noise like button autorepat.

fruit-bat, 2024-08-21:
You might need the latest sdk... but try completely clearing your build folder and running cmake again.  Make sure you have my version of pico extras.

DPRCZ, 2024-08-22:
OK, I'll test your .UF2 file to ensure that problem is on my setup.  And I let you know the result.

DPRCZ, 2024-08-22:
With your "ZxSpectrumPicomputerVgaAukBob_640x480x60Hz" is VGA working. Issue is on my PC . Letters  are not white but light blue now on Menu?

fruit-bat, 2024-08-23:
> With your "ZxSpectrumPicomputerVgaAukBob_640x480x60Hz" is VGA working. Issue is on my PC . Letters are not white but light blue now on Menu?

Yup, VGA is working but the menu has gone a rather nice blue colour... don't know why. I will have a look.

fruit-bat, 2024-08-23:
I've updated pico-dvi-menu to better handle the difference VGA encodings. Also, the builds can define the menu colour with symbol, e.g.

  PICO_MENU_FOREGROUND_COLOR=COLOR_BRIGHT_GREEN

Now defaulting to bright-white on black.

DPRCZ, 2024-08-24:
My own compilation works  again. I've refreshed all from scratch from github.

DPRCZ, 2024-08-24:
Fixed.

closed 2024-08-24 by DPRCZ