fruit-bat/pico-zxspectrum #168: 8-bit NES/SNES Gamecontroller support ?

javavi, 2024-08-24:
The MURMULATOR platform widely uses 8-bit joysticks from NES clones due to the simplicity of the connection protocol and their cheapness on AliExpress. One joystick is connected via 3 wires, and two joysticks together via 4 wires. The latest boards have a DB9 connector for connecting them (although this is not an original NES connector).
New NES controllers operate on 3.3V Pico power and do not require level matching.
Is it possible to support such game controllers here?

javavi, 2024-08-24:
[image]

fruit-bat, 2024-08-25:
https://gamefaqs.gamespot.com/snes/916396-super-nintendo/faqs/5395

javavi, 2024-08-26:
https://tresi.github.io/nes/
https://www.bertiusgames.com/projects/gamepad.html

Example of NES/SNES controller driver code from one firmware for Murmulator:
https://github.com/xrip/pico-nes/tree/main/drivers/nespad

fruit-bat, 2024-08-26:
Yup, something like...
> .program nespad
> 
> .side_set 2
> .wrap_target
>     nop side 0b10 [7]
>     nop side 0b10 [7]
>     set x, 16 side 0b00 [5]
> loop:
>     in pins, 2 side 0b00 [1]
>     nop side 0b01 [7]
>     jmp x--, loop side 0b00 [5]
> .wrap

fruit-bat, 2024-08-26:
https://wokwi.com/projects/407292544456785921

fruit-bat, 2024-08-26:
[image]

fruit-bat, 2024-08-27:
https://wokwi.com/projects/407394757834126337

javavi, 2024-08-31:
[image]

fruit-bat, 2024-09-02:
I'm waiting on some sockets to try out my emulated code. They should turn up in a few days.

fruit-bat, 2024-09-04:
[image]

javavi, 2024-09-07:
The original "old" NES/SNES gamepads may not work with a supply voltage of 3.3V. They need to be supplied with 5V and, accordingly, coordinated with the Pico TTL logic 3.3V at least for the DATA line.
  Also, their proprietary NES/SNES connectors are hard to get, so our enthusiasts prefer to use NES controllers with DB-9 connector from unofficial clones.

fruit-bat, 2024-09-07:
I've got a couple of these unbranded remakes. Hopefully, they will run from 3.3v! In theory, the 2040 pads will cope with 5v, so I could try running the pads at 5v and leave the signalling at 3.3v.

[image]

fruit-bat, 2024-09-07:
Just given them a try and they work of 3.3v (phew!). Beginnings of the driver here...
https://github.com/fruit-bat/pico-nespad

The idea is to continuously sample using a PIO state-machine, draining the FIFO using an ISR to update a 'register'.

The code needs tidying up and turning into a library with example app... but it is looking promising.

fruit-bat, 2024-09-09:
Library and demo application now working nicely :-)

https://github.com/fruit-bat/pico-nespad

Will try to add it to the emulator when I get chance.

fruit-bat, 2024-09-13:
Please would you let me know which, if any, of these work...
[ZX-MURMULATOR.uf2.zip](https://github.com/user-attachments/files/16998491/ZX-MURMULATOR.uf2.zip)

(If the first few fail might not be worth going through them all)

javavi, 2024-09-14:
> Please would you let me know which, if any, of these work... 
I tried several firmwares, they work, but unfortunately the NES gamepad does not respond. 
The NES gamepad is connected according to the following working scheme 
[image]

fruit-bat, 2024-09-14:
Do you have an oscilloscope or anything you can use to check if the clock and enable signals are working? A multimeter with a frequency counter would probably do.

Choose one of the firmware, and let's focus on getting one of them functional. Let me know which you choose and can test.

fruit-bat, 2024-09-14:
... I think I have spotted a mistake. I will do a new build and post a new attempt shortly....

fruit-bat, 2024-09-14:
Less ambitious attempt, VGA with PWM audio...

[ZX-MURMULATOR_VGA_PWM_AUDIO_720x576x50Hz.uf2.zip](https://github.com/user-attachments/files/17002168/ZX-MURMULATOR_VGA_PWM_AUDIO_720x576x50Hz.uf2.zip)

javavi, 2024-09-14:
> Less ambitious attempt, VGA with PWM audio...
Yes! My NES gamepad works on this firmware, but the mapping to Kempston port is strange: A00BUDLR
The SEL and START buttons are not mapped

javavi, 2024-09-14:
In many Spectrum emulator firmwares using the NES controller, a modern 8-bit standard for mapping data to the Kempston port of the joystick has been developed.
`
A  B  Se St U  D  L  R <-- NES controller
7  6  5  4  3  2  1  0
St Se A  B  U  D  L  R --> Kempston port`

fruit-bat, 2024-09-14:
This better...

[ZX-MURMULATOR_VGA_PWM_AUDIO_720x576x50Hz.uf2.zip](https://github.com/user-attachments/files/17002871/ZX-MURMULATOR_VGA_PWM_AUDIO_720x576x50Hz.uf2.zip)

?

javavi, 2024-09-14:
> This better...
Yes! Now only button B doesn't work at all. 8))

fruit-bat, 2024-09-14:
Hmm... maybe your joysticks work different to mine... it works ok for me.

Is the order of your buttons like mine?

<code>
  U             X
L   R  Se  St  Y  A   
  D             B
</code>

```
#define NESPAD_BI_B 0
#define NESPAD_BI_Y 1
#define NESPAD_BI_SELECT 2
#define NESPAD_BI_START 3
#define NESPAD_BI_UP 4
#define NESPAD_BI_DOWN 5
#define NESPAD_BI_LEFT 6
#define NESPAD_BI_RIGHT 7
#define NESPAD_BI_A 8
#define NESPAD_BI_X 9
#define NESPAD_BI_LS 10
#define NESPAD_BI_RS 11
```

fruit-bat, 2024-09-14:
Perhaps I need top swap the following?

| mine | yours |
| - | - |
| Y | B |
| B | A |

javavi, 2024-09-14:
> Hmm... maybe your joysticks work different to mine... it works ok for me.

That's right!
I have an 8-bit NES joystick (even though it looks like SNES)
You have a 12-bit SNES joystick.

fruit-bat, 2024-09-14:
> > Hmm... maybe your joysticks work different to mine... it works ok for me.
> 
> That's right! I have an 8-bit NES joystick (even though it looks like SNES) You have a 12-bit SNES joystick.
Yes, but we only need the lower 8 bits for the mapping you requested.

[ZX-MURMULATOR_VGA_PWM_AUDIO_720x576x50Hz.uf2.zip](https://github.com/user-attachments/files/17002957/ZX-MURMULATOR_VGA_PWM_AUDIO_720x576x50Hz.uf2.zip)

fruit-bat, 2024-09-14:
I may still have 8bit NES A & B round the wrong way. Let me know if they need swapping.

javavi, 2024-09-14:
The 'B' button still doesn't work on my NES game controller  :(

javavi, 2024-09-14:
> I may still have 8bit NES A & B round the wrong way. Let me know if they need swapping.

The 'B' button still doesn't work on my NES game controller :(

fruit-bat, 2024-09-14:
I think the first one should work (but I'm getting to tired to think straight, so I have put the reverse mapping after it):

[ZX-MURMULATOR_VGA_PWM_AUDIO_720x576x50Hz_BY.uf2.zip](https://github.com/user-attachments/files/17003008/ZX-MURMULATOR_VGA_PWM_AUDIO_720x576x50Hz_BY.uf2.zip)
[ZX-MURMULATOR_VGA_PWM_AUDIO_720x576x50Hz_YB.uf2.zip](https://github.com/user-attachments/files/17003010/ZX-MURMULATOR_VGA_PWM_AUDIO_720x576x50Hz_YB.uf2.zip)

javavi, 2024-09-14:
> ZX-MURMULATOR_VGA_PWM_AUDIO_720x576x50Hz_BY.uf2.zip

The firmware version _BY for the NES game controller works as it should.

fruit-bat, 2024-09-15:
Now for an HDMI build...

[ZX-MURMULATOR_HDMI_HDMI_AUDIO_720x576x50Hz.uf2.zip](https://github.com/user-attachments/files/17006541/ZX-MURMULATOR_HDMI_HDMI_AUDIO_720x576x50Hz.uf2.zip)

javavi, 2024-09-15:
> Now for an HDMI build...

Excellent! NES gamepad works as it should!

javavi, 2024-09-17:
No sound in firmware with _HDMI_I2S_AUDIO_***
In the rest everything is OK! 
NES joystick works correctly everywhere.

fruit-bat, 2024-09-18:
Cheers, these any better?

[ZX-MURMULATOR-I2S.zip](https://github.com/user-attachments/files/17038810/ZX-MURMULATOR-I2S.zip)

javavi, 2024-09-18:
> Cheers, these any better?

Now it's OK! :)

closed 2024-09-27 by javavi