fruit-bat/pico-zxspectrum #169: PicoDVI, SDK 2.0, RP2350

ikjordan, 2024-08-26:
I've updated the audio branch of my PicoDVI fork so that it compiles with SDK 2.0. I've also checked that it works with Pico2 (i.e. RP2350). It does not use any of the new features (such as the SIO TMDS encoder or HSTX) but the same code base does run on both RP2040 and RP2350 (as long as the M33 cores are used, rather than Hazard3).

I've successfully run picozx81 with a pico2 on the Pimoroni and Olimex carrier boards, so hopefully it should not be too difficult to port pico-zxspectrum to the RP2350. All that is left then is to decide what to do with all of the extra memory and CPU capacity ;-)

fruit-bat, 2024-08-26:
Very nice. Thanks for the update... the new HSTX interface looks tempting.

ikjordan, 2024-09-03:
That's great. It definitely runs much faster :-)

closed 2024-09-03 by ikjordan