fruit-bat/pico-zxspectrum #170: Add a key combination to trigger BOOTSEL?

ikjordan, 2024-09-03:
As you know, to update to new firmware it is necessary to go into BOOTSEL mode. Typically this is done by pressing the BOOTSEL button, but that can be tricky for cased PCBs.

I wonder if you would consider adding a key combination that would reset the board into BOOTSEL mode? That may make swapping and upgrading uf2 updates a little easier. 

As an example in picozx81 when running on the [PICO ZX48/128](https://github.com/fruit-bat/pico-zxspectrum/blob/main/docs/pico_zx48_128.md) it checks for the menu key being pressed at the same time as the R key. If both are pressed a reboot into BOOTSEL mode is made through a call to the SDK function `reset_usb_boot(0,0)`   
[code link](https://github.com/ikjordan/picozx81/blob/c452bd7ae0a613f30c004b60c3a95baec2ff31ba/src/emukeyboard.cpp#L162)

fruit-bat, 2024-09-04:
Might add it to the menu, unless you think that is a terrible idea?

ikjordan, 2024-09-04:
Using the menu sounds a better idea than mine :-) If you could add that it would be great.

One other suggestion / request whilst I'm at it. Would it be possible to add support for the [Olimex RP2040-PICO-PC](https://thepihut.com/products/olimex-rp2040-pico-pc-for-raspberry-pi-pico)? It is quite inexpensive and works nicely. The fork of PicoDVI already has the necessary additions for it. Despite the name it works with the RP2350. It also has the correct HDMI pinout (12 to 19) to support HSTX, I got the Pico HSTX SDK example code running on it with some trivial changes. I suspect it might sell well in the next few months!

fruit-bat, 2024-09-04:
I could have a go with the Olimex... I don't have one so making sure it works will be tricky.

How did you get the HSTX example working... I don't get a stable image... lots of horizontal banding :-(

fruit-bat, 2024-09-04:
https://github.com/fruit-bat/pico-zxspectrum/pull/171

Thanks @ikjordan. I wish I had done this a long time ago!

ikjordan, 2024-09-04:
> Thanks @ikjordan. I wish I had done this a long time ago!

That's great!

I got the HSTX [dvi_out_hstx_encoder example](https://github.com/raspberrypi/pico-examples/tree/master/hstx/dvi_out_hstx_encoder) running on the Olimex board by swapping the clock and D0 lines and changing the inversions (lines 196 to 219). I posted the code changes  [here](https://www.thebackshed.com/forum/ViewTopic.php?TID=17187&P=2)

However, looking at the pinouts for ZxSpectrumBreadboardHdmiNPinAudio and ZxSpectrumBreadboardHdmiKbd1PinAudio, it seems like dvi_out_hstx_encoder should work with those boards out of the box if a RP2350 is fitted? All the other HDMI boards seem to have the wrong pinouts.

One thing to note is that the example assumes that the default clock is 125MHz, whereas it is really 150MHz, so the example runs at a 72Hz refresh rate, rather than 60Hz. I wonder if that could have upset your monitor?

ikjordan, 2024-09-05:
> I could have a go with the Olimex... I don't have one so making sure it works will be tricky.

I'm happy to try out any pico-zxspectrum test binaries with the Olimex board, if that would help

fruit-bat, 2024-09-07:
@ikjordan 

> > I could have a go with the Olimex... I don't have one so making sure it works will be tricky.
> 
> I'm happy to try out any pico-zxspectrum test binaries with the Olimex board, if that would help

See :
https://github.com/fruit-bat/pico-zxspectrum/issues/172
https://github.com/fruit-bat/pico-zxspectrum/tree/feature/olimex-pc

Please let me know what works in the two builds (if anything).

Cheers :-)

DPRCZ, 2024-09-07:
Hm, firmware update is nice feature but switch to usb bootloader is not very usefull when your raspberry pico is inside zx spectrum plastics box.  I would preffer update via SD card -> When system find firmware.bin on root of sdcard at the booting /start, System will update flash from this file and then delete the file or rename.

ikjordan, 2024-09-07:
I found it works well when using a Pico on the Bobricius board inside a Spectrum case. The key point, is that there is a keyboard controlled over GPIO but the device is powered by a USB cable with data connections to the Pico. I've used this approach to update the uf2 in such a system.

I agree that it is not much use unless there is a non USB keyboard attached, as it is not possible to update a uf2 via a hub that also has a keyboard attached. So if you have a board inside a plastic case, but then have to control it with an external USB keyboard then this change probably does not help.

I'm just finishing testing the changes above - I'll report back in a few mins.

DPRCZ, 2024-09-07:
So you  use USB A - USB A cable to firmware update?

ikjordan, 2024-09-07:
Yes - on the Bobricius  board I use USB A to USB A to power and update

ikjordan, 2024-09-07:
Tested the Olimex changes successfully, but during testing hit an issue that is not Olimex related. Raised #174

ikjordan, 2024-09-20:
Submitted and working, so closing the issue as completed

closed 2024-09-20 by ikjordan