fruit-bat/pico-zxspectrum #172: Build for Olimex RP2040-PICO-PC

fruit-bat, 2024-09-07:


fruit-bat, 2024-09-07:
https://github.com/OLIMEX/RP2040-PICO-PC/blob/main/HARDWARE/RP2040-PICO-PC%20hardware%20revision%20B/RP2040-PICO-PC_rev_B.pdf

ikjordan, 2024-09-07:
I've tested this on an Olimex board. DVI and SD Card is working at both resolutions. So this is good to merge.

ikjordan, 2024-09-07:
Incidentally, when building this from source I had to go through the step where I moved the default tinyusb out to use the specific version for pico-zxspectrum.
When I started on picozx81 I "borrowed" the usb code from this project, but factored out the extra joystick code, so I could still use the tinyusb that is in the SDK. That may make SDK updates easier. Feel free to borrow that code back :-)

closed 2024-09-07 by fruit-bat

fruit-bat, 2024-09-09:
@ikjordan I will definitely have a look at how you did that. Originally, I though I needed to change TinyUSB to 'hook' my changes in but maybe that is not the case anymore (or never was!).