fruit-bat/pico-zxspectrum #175: Move joystick/mouse code out of TinyUSB

fruit-bat, 2024-09-09:
See https://github.com/ikjordan/picozx81/tree/main/usb

ikjordan, 2024-09-11:
I see the pico-zxspectrum code has moved on a bit since I copied it (e.g. the addition of xinput).
The changes I made were to move hid_ri.c, hid_rip.c hid_ri.h and rid_rip.c out of TinyUSB and then modify the application code to point to the new file locations. 
If you do this change I'll look forward to merging your code back into picozx81 and automatically getting support for some more HID devices :-)

fruit-bat, 2024-09-15:
Just copied some files in, but is a start...

https://github.com/fruit-bat/pico-hid-host

fruit-bat, 2024-09-22:
I have a working test app and some tests that pass. Needs some work to make it a bit more friendly but interested in what you think of it.

https://github.com/fruit-bat/pico-hid-host

I will try adding to the spectrum next.

ikjordan, 2024-09-29:
Hi sorry for not replying sooner. Thank you for this. I don't have access to any test kit for a while, but will look at in more detail when I do.