fruit-bat/pico-zxspectrum #176: noobiest of noob questions

jonnypoko, 2024-09-13:
Hi, stupid question about setup - I have built two of these this week (Bobricius boards with and without integrated keyboards) with the intention of presenting it to a friend for his 50th Birthday tomorrow! so I am rushing and running out of time for troubleshooting - which is probably why I have missed this (I usually like to persevere more with snagging before asking for help) Both boards boot into the spectrum 128k menu screen but the on-board cursor and keyboard do nothing - I have been able to get into the emulator menu to confirm the circuits havenent frozen, but there is no movement of the cyan bar on the sinclair menu - is this normal? i.e. is just a splash screen and everything is done through the emultor menu or am I missing a setup step? I'm sure this is answered elsewhere - just point me at the 'FAQ for dummies' cheers!!

fruit-bat, 2024-09-13:
Post a pic of the boards just so I can be sure which you have.

fruit-bat, 2024-09-13:
Tell me which firmware you are using and if you got it from this site

DPRCZ, 2024-09-13:
Could be issue with keyboard connectors / swapped side or if you use old membrane keyboard it could be already dead. First of all try to attach USB keyboard.

DPRCZ, 2024-09-13:
[image]

jonnypoko, 2024-09-13:
[image]

jonnypoko, 2024-09-13:
Sorry, what a nuts day - just got free, So the keys that I expected to work for the spectrum menu were the discrete cursor buttons in the circle or the keyboard buttons on the top board that I have soldered - plugging in a usb keyboard works - I havent tried a membrane yet

jonnypoko, 2024-09-13:
Firware is "ZxSpectrumPicomputerVgaAukBob_720x576x50Hz.uf2" from this site yes

DPRCZ, 2024-09-14:
"ZxSpectrumPicomputerVgaAukBob_720x576x50Hz.uf2" this firmware could be tricky for moust VGA monitor, but if you have picture  on screen - it works for you.

Try firmware from this page: 
https://www.pcbway.com/project/shareproject/PICOZX_motherboard_for_ZX_Spectrum_original_case_5bbde8be.html

fruit-bat, 2024-09-14:
I had to remind myself how that all worked, so I updated the document here:
https://github.com/fruit-bat/pico-zxspectrum/blob/main/docs/pico_zx48_128.md

Note that you should be able to move up and down the menu using CAPS SHIFT 7 and CAPS SHIFT 6 respectively.

If you find other things you wish had been in the documentation please let me know and I will add them. 

Hope the birthday celebrations go well!

#### Joystick buttons & modes
The PCB has some buttons which can act as a joystick or the ZX Spectrum cursor.
The LED on the Pico PI indicates the mode:

| LED | mode |
| - | - |
| off | joystick mode |
| on | cursor mode |

To switch mode make sure you are viewing the ZX Spectrum display (not the menus),
then press CAPS SHIFT and one of  <<< or >>> at the same time:

| Keys | Mode |
| - | - |
| CAPS SHIFT <<< | Cursor |
| CAPS SHIFT >>> | Joystick |

When the machine boots, the buttons are configure to act as a joystick.

When in joystick mode the buttons act as a Kemptson joystick.

jonnypoko, 2024-09-15:
Thanks guys - I'll try all that