fruit-bat/pico-zxspectrum #179: Pico2 RP2350 SDK2.0 TinyUSB17.0

javavi, 2024-09-27:
Now users are starting to see the new Pico2 on the RP2350 chip.

Is it possible to organize the code here so as to also get the firmware for the ZX-Spectrum emulator for the new Pico2?

fruit-bat, 2024-09-28:
Yup, in theory everything should build for the 2350 already, but I am currently looking at moving some of my changes I made in TinyUSB into a separate library... which *should* make tracking changes to TinyUSB easier and allow version 17 to be used.

Do you think we need pre-built firmware for the RP2350? I kind of assumed until Raspberry PI promise to fix the pads (which seems unlikely as they seem to be implying it is all just fine, when clearly it is not) it was not really worth investing too much time in.

javavi, 2024-09-28:
> Do you think we need pre-built firmware for the RP2350?

I think that the defect of the RP2350 chip does not affect us much, but those advantages are very attractive. In the Murmulator user community, many have already purchased the new Pico2 to try. Yes, rebuilt projects on the new Pico 2 are already appearing, especially those that lacked resources, such as the Sega MegaDrive 2 emulator, etc., are winning.
https://marketplace.visualstudio.com/items?itemName=raspberry-pi.raspberry-pi-pico

javavi, 2024-09-28:
VSCode extension 
Seamlessly switch between different versions of the Pico SDK and tools.
https://marketplace.visualstudio.com/items?itemName=raspberry-pi.raspberry-pi-pico

ikjordan, 2024-09-29:
I'd agree that the RP2350 pad issue should not impact this project. I think all of the keyboard code uses pull-ups, which are not impacted, rather than pull-downs? The Olimex board testing I did with the RP2350 worked fine.
In theory it is possible to build one uf2 that includes code for both MCU types and the boot loader then automatically picks the right one, but I've not tried that personally.

fruit-bat, 2024-09-29:
A combined binary would be nice. I'll see if I can find out how to make one.

fruit-bat, 2024-09-29:
> I'd agree that the RP2350 pad issue should not impact this project. I think all of the keyboard code uses pull-ups, which are not impacted, rather than pull-downs? The Olimex board testing I did [...]

I've tried just concatenating the uf2 files from the RP2040 & RP2350 builds, the result of which seems to work fine on both boards!

ikjordan, 2024-09-29:
That's excellent! Something else I'll be able to "borrow" from this project in the future ;-)

fruit-bat, 2024-09-29:
@javavi  here are some combined builds...

[ZX-MURMULATOR.zip](https://github.com/user-attachments/files/17179459/ZX-MURMULATOR.zip)

javavi, 2024-09-29:
> @javavi here are some combined builds...

Yes, I work on both Pico1 and Pico2, but not all of them! ;)
On some, when booting the PC, it complains about the lack of memory, but then they work fine...

fruit-bat, 2024-09-29:
> How do you get such combined firmware files? Is it with the help of that extension for VSCode?

Used 'cat' command in Linux.  Probably can do something similar with copy in windows.

fruit-bat, 2024-09-29:
> > @javavi here are some combined builds...
> 
> Yes, I work on both Pico1 and Pico2, but not all of them! ;)
> For example, it doesn't work ZX-MURMULATOR_VGA_PWM_AUDIO_720x576x50Hz.uf2  on Pico2, although it works on Pico1.
> On some firmware, when booting the Windows PC, it complains about the lack of memory, but then they work fine...
> 
> 

Hmm, I  wonder if it matters if the rp2040 binary comes first.

fruit-bat, 2024-09-29:
Possible the 2350 builds don't all work! Here are just the plain 2350 builds...

[ZX-MURMULATOR-2350.zip](https://github.com/user-attachments/files/17179784/ZX-MURMULATOR-2350.zip)

javavi, 2024-09-29:
> Possible the 2350 builds don't all work! Here are just the plain 2350 builds...

So all these firmwares work on Pico 2 (RP2350) on Murmulator

fruit-bat, 2024-10-01:
Thanks for testing them.... perhaps combining builds for pico 1 & pico 2 is not is not reliable. 

I will keep look at improving the cmake system to make building sets of firmware easier.

javavi, 2024-10-12:
> Thanks for testing them.... perhaps combining builds for pico 1 & pico 2 is not is not reliable.
Everything is fine!
It's just that when writing the firmware, if it comes first in the file, the OS gives an error, since the file copying will not be completed. However, the necessary firmware is written normally.

fruit-bat, 2024-10-12:
@javavi  thanks, I'm still struggling a bit organising the cmake stuff... somewhat regretting my crazy naming of the firmware... and along with not really knowing cmake well enough.

fruit-bat, 2024-10-12:
If you think what I am doing on the feature/cm-rp2350 branch is crazy do tell me.

javavi, 2024-10-12:
> If you think what I am doing on the feature/cm-rp2350 branch is crazy do tell me.

In one project I am following, everything is contained in one place and only the СMake option specifies which chip to build the firmware for.
https://github.com/UKTailwind/PicoMiteAllVersions
On the other hand, separating the repositories will simplify the development of the code with the capabilities of the new RP2350 chip.

closed 2025-01-11 by javavi