fruit-bat/pico-zxspectrum #182: Exposing Pico GPIO pins

gpg342, 2024-10-22:
Upfront - thanks for the excellent work on this project.

Not sure if this is a realistic question....

Sort of related to the thread on PicoMites / MMBasic for STEM projects...

Is it possible to get access to some (any) of the unused GPIO pins on the Pico via the emulator?  Looking to be able to wire a couple of LEDs up and make them flash etc -- and possibly detect input via an analogue pin?   Could this be achieved with a PEEK / POKE combination?

The idea being that in class I can use the Pico-Spectrum as a speccy and then drop into 128k BASIC and "do some stuff" like code a set of traffic lights or read the value of an LDR.

The above is what I would use MMBASIC for and it works - but of course MMBASIC is not a speccy!!.

Regardless - kudos and keep up the excellent work,

Glen

fruit-bat, 2024-10-23:
Sounds possible, particularly for the digital ouputs. Would you be happy with 'out' on port(0FFD), rather than 'poke'. Which gpio on the Pico do you want to use for the leds? I'll have a think about the analogue input.

fruit-bat, 2024-10-23:
...made a start.

gpg342, 2024-10-23:
Sorry for the late reply - long day in school.

Any free GPIO would work - let me look at the board specifically.   Out to a port address would be perfect.

For what it's worth - this would open up the Pico-ZX to the world of education in a big way - firstly the "cost" for a Boots to basic computer (like MMBasic / PicoMite) + the allure of actually playing real games from an extensive library would make this a nice piece of kit.

The limit of the intended use case would be to drive some LEDs and possibly discrete logic - to simulate things like traffic lights.  Input would open this further to interactivity - sense the light / read a thermister --- nothing as clever as I2C etc as this is beyond school curricular - and to be honest beyond what could reasonably be coded without a library of some sort -- keep it simple ;-)

Appreciate the come back!

Gx

fruit-bat, 2024-10-25:
Does the pin usage from here match the boards you are using?

https://land-boards.com/blwiki/index.php?title=PicoVGA01

fruit-bat, 2024-10-25:
Give this a go...

4 digital outputs on GP10, GP11, GP12, GP13 and one analogue input on GP28

To read the analogue input:
<pre>in 4095</pre>
...which will result in a number from 0 to 255, corresponding to 0v to 3.3v.

To set the digital outputs, output to the same port with 
D0 - GP10
D1 - GP11 
D2 - GP12
D4 - GP13 

e.g.  set GP10 and GP13
<pre>out 4095, 9</pre>

[ZxSpectrumPicoMiteVGA_720x576x50Hz.zip](https://github.com/user-attachments/files/17526880/ZxSpectrumPicoMiteVGA_720x576x50Hz.zip)

(remember to unzip the uf2 from the zip file)

gpg342, 2024-10-29:
Giving this a go -- on  holiday (half term in school - teacher) -- so out and about -- If I go quiet for a couple of days -- I'll get onto testing when I get back.   Gx

fruit-bat, 2024-10-29:
Thanks for letting me know and hope you have a good holiday.  I'm happy you are planning to use the project in your class; let me know how it goes 🙂 and if you need further updates.

gpg342, 2024-11-01:
Back at it now - thanks for the patience.  My board is currently a Pi-Zero based setup -- will that effect this OR do I need to build up a different board?

Sorry for the daft question...

Gx

fruit-bat, 2024-11-01:
Yup, the different boards need different firmware.  Does your look like the on on this page...

https://github.com/fruit-bat/pico-zxspectrum/blob/main/docs/ZxSpectrumPiZero.md

If not post a pic of it.

Then, if I get time over the weekend, I'll do a new build and post it here.

gpg342, 2024-11-01:
Yes, thats the puppy!!

Gx

fruit-bat, 2024-11-02:
Schematic: https://files.waveshare.com/wiki/RP2040-PiZero/RP2040-PiZero-Schematic.pdf

From looking at the circuit diagram, it appears that all 4 ADC pins are routed to the DVI port, for the display. This is such a strange thing to do, I keep checking the diagram, but it does seem to be the case.

It appears,  gp0 to gp4 and gp7 to gp15 are free. They could be used for digital io, if you think it has enough purpose without the ADC?

gpg342, 2024-11-02:
Arrh - yes, from the schematic https://files.waveshare.com/wiki/RP2040-PiZero/RP2040-PiZero-Schematic.pdf  it appears so -- odd choice from Waveshare.   It also shows GPIO 7/8 linked to "CE0" and "CE1" internally.

From the example code it looks like the "negative" signals are derrived from just the positve by inverting them - so 24/26/28 are used but 25/27/29 are not directly accessed (but are routed to the HDMI header) -- but my schematic / electronics kung-fu might be limited.  

Digital I/O would be a start - and really useful.  It might be better for me to build up another board with the ADC available., I'll have a look for the simplest (making one for me is fine, but for use in class the most robust is usually one that is integrated from the the get go)  -- I'll have a look and see if I can make up the "easiest" variant.

Cheers
G

gpg342, 2024-11-02:
^^^ I mean I could possibly cut traces that route 25/27/29 to the HDMI port.

(Happy to see about about building up another - but the Waveshare as is has SD / USB etc out of the box).

Regardless - I really appreciate your input on this and regardless the project will see use in class.

Gx

fruit-bat, 2024-11-02:
I think cutting the tracks will break the dvi ouput. I believe the rp2040
creates the differential signal.

fruit-bat, 2024-11-02:
Might be worth looking at which pins are free on this board?

https://thepihut.com/products/olimex-rp2040-pico-pc-for-raspberry-pi-pico

gpg342, 2024-11-02:
MMmm - so many of these board out there.

No dice -- whilst not all routed to DVI, some the rest are routed to the PWM for the audio -- but could cut those and remove the audio jack.

But - a nice board, will snag one.

Gx

Honestly - dont sweat this too much -- my use case is very much an "edge" case -- the Waveshare with digital I/O will suffice for a proof of concept and some LED action in class.

Gxx

fruit-bat, 2024-11-02:
<pre>
// Output pins on GP9, GP10, GP11, GP12
static const uint8_t POUT_PINS[] = {9, 10, 11, 12};

// Input pins on GP13, GP14, GP15
static const uint8_t PIN_PINS[] = {13, 14, 15};
</pre>

[ZxSpectrumPiZero_720x576x50Hz.uf2.zip](https://github.com/user-attachments/files/17607057/ZxSpectrumPiZero_720x576x50Hz.uf2.zip)

I don't have one of these boards, so I'm afraid I don't know if this works. If it even boots, it is a start.

To read the digital inputs:
<pre>in 4095</pre>

To set the digital outputs, output to the same port with 
D0 - GP9
D1 - GP10 
D2 - GP11
D3 - GP12 

e.g.  set GP10 and GP13
<pre>out 4095, 9</pre>

gpg342, 2024-11-02:
Perfect - will give it a go.

IIRC the out 4095,9 writes binary 1001  which is pins 9 and 12 in your example?

Will have a play over the weekend.

Much thanks for trying this.

Gx

fruit-bat, 2024-11-02:
> Perfect - will give it a go.
> 
> IIRC the out 4095,9 writes binary 1001 which is pins 9 and 12 in your example?
> 
> Will have a play over the weekend.
> 
> Much thanks for trying this.
> 
> Gx

Yes, that's correct