fruit-bat/pico-zxspectrum #185: LCD Display not working

cosmicfrog3000, 2024-11-12:
Hi. I cannot get the LCD display working (ST7789 and ILI9341) All I get is a white screen. I think the problem may be with the RESET line. Should this be connected? If so where?
Many thanks.
Dave.

cosmicfrog3000, 2024-11-12:
Got it working!
The reset pin goes to the RUN pin of the pico.
Dave.

fruit-bat, 2024-11-12:
Glad you got it working!

cosmicfrog3000, 2024-11-12:
I'm wanting to use this on a RP2040 ZERO Waveshare board but they don't have a RUN Pin. It looks like you can just connect the RESET to the 3.3v line and it will work. 
Would it be ok and safe to do this?
Many thanks.
Dave.

cosmicfrog3000, 2024-11-12:
Thanks for the reply. I've tried it on a standed RP Pico and it works ok, but I'll try it with a resistor to be safe.
Great emulator by the way!
Dave.