fruit-bat/pico-zxspectrum #189: New platform MURMULATOR2

javavi, 2024-11-23:
A new version of the MURMULATOR2 platform and the first revision of the HW board ver.2.0 have been released.
The entire rearrangement is for the sake of using the HSTX pins RP2350 of the HDMI video output.
Although this platform can be used with both the new Pico2 RP2350 and the previous Pico1 RP2040.
[image]
[image]
Can I ask you to create a firmware configuration for this new MURMULATOR2 platform?

fruit-bat, 2024-11-24:
Very nice board! 

Yep, I will have a go at adding builds for it.

fruit-bat, 2024-11-24:
Any life in this lot?

[MURM2.zip](https://github.com/user-attachments/files/17894193/MURM2.zip)

javavi, 2024-11-24:
> Any life in this lot?

No, unfortunately it didn't work right away...!
This is how I understood it, there were firmwares for Pico1 RP2040

fruit-bat, 2024-11-24:
I'm living in the past! Here are some RP2350 builds. At the very least you should get screen.

[MURM2_RP2350.zip](https://github.com/user-attachments/files/17894539/MURM2_RP2350.zip)

javavi, 2024-11-24:
No, it didn't work again! Maybe it will work the third time!
It is not absolutely necessary to install a new Pico2 on the Mulmulator2 board, it can also be used for the first Pico, and although it was made with HDMI in mind, it is possible to have VGA on it, I have an adapter for this.

fruit-bat, 2024-11-24:
...ah, I think my dvi sock is wired differently to your hdmi port... looks like clk and d0 are swapped. Looks like pairs are swapped.  Looks similar to the olimex board... I will try that...

fruit-bat, 2024-11-24:
static const struct dvi_serialiser_cfg Olimex_RP2040_PICO_PC_cfg = {
	.pio = pio0,
	.sm_tmds = {0, 1, 2},
	.pins_tmds = {14, 18, 16},
	.pins_clk = 12,
	.invert_diffpairs = true
};

[Uploading MURM2_RP2350_v2.zip…]()

fruit-bat, 2024-11-24:
[MURM2_RP2350_v2.zip](https://github.com/user-attachments/files/17894693/MURM2_RP2350_v2.zip)

javavi, 2024-11-24:
It worked!
So far only video, oh there is a problem with the color palette, the blue color here is purple, the PS/2 keyboard does not work, it is now on the GP2, GP3 pins

fruit-bat, 2024-11-24:
Possibly fixed colours:

> static const struct dvi_serialiser_cfg murmulator2_hdmi_cfg = {
> 	.pio = pio0,
> 	.sm_tmds = {0, 1, 2},
> 	.pins_tmds = {14, 16, 18},
> 	.pins_clk = 12,
> 	.invert_diffpairs = true
> };

Possibly fixed PS/2 keyboard.

[MURM2_RP2350_v3.zip](https://github.com/user-attachments/files/17894783/MURM2_RP2350_v3.zip)

javavi, 2024-11-24:
Oops! This broke everything, now the screen is black.
I thought it would be simple, just changing the pin numbers in the definitions.

fruit-bat, 2024-11-24:
Not sure why that resulted in a blank screen.  Might be cos it tried to use the HSTX (the audio was added before the HSTX)... this is build against an older version.

[MURM2_RP2350_v4.zip](https://github.com/user-attachments/files/17895016/MURM2_RP2350_v4.zip)

(on the diagram the pins have moved, but inside the pico quite a lot has changed)

javavi, 2024-11-25:
There is some video signal, but no image, black screen.
Are you trying to use the HSTX Pico2 module?

fruit-bat, 2024-11-26:
I think the latest version of PicoDVI uses it if the right pins are selected... hence trying an older version.  When I get time I will wire up an hdmi connector the same as your board so I can debug.

fruit-bat, 2024-12-01:
HDMI config is fine. PS/2 keyboard state-machine is flooding it's input buffer which is causing the emulator to stall.

fruit-bat, 2024-12-01:
[MURM2_RP2350_v5.zip](https://github.com/user-attachments/files/17969680/MURM2_RP2350_v5.zip)

javavi, 2024-12-01:
Yes! It's OK! 
I checked, HDMI video and sound, PS/2 keyboard work.
You can output the printer output, LLIST and LPRINT commands to UART.  I'm sure some people will like it very much.

fruit-bat, 2024-12-08:
Here is VGA build. Please let me know what if anything works...

[MURM2_VGA_I2S_RP2350_v1.zip](https://github.com/user-attachments/files/18051286/MURM2_VGA_I2S_RP2350_v1.zip)

javavi, 2024-12-08:
Yes! It's work on Raspberry Pi Pico2 RP2350!
VGA video and I2S sound, SD card work, PS/2 keyboard work, NES gamepad work, TAPE LOADING also works.

javavi, 2024-12-08:
In the menu, the second line, somehow it seems to me that it is written in the wrong order.
I would suggest the following:
**ZX-Spectrum 48K/128K by fruit-bat on Murmulator2
Powered Raspberry Pi Pico RP2040 @250.0MHz****
or
**ZX-Spectrum 48K/128K by fruit-bat on Murmulator2
Powered Raspberry Pi Pico2 RP2350A @250.0MHz**
respectively.

fruit-bat, 2024-12-08:
> Yes! It's work on Raspberry Pi Pico2 RP2350! VGA video and I2S sound, SD card work, PS/2 keyboard work, NES gamepad work, TAPE LOADING also works. I would like to have on the MURMULATOR2 board not only firmware for Pico2 RP2350 but also all the traditional ZX-Spectrum firmware from FruitBat for Pico1 RP2040

[MURM2_RP2040_all_v1.zip](https://github.com/user-attachments/files/18051589/MURM2_RP2040_all_v1.zip)
[MURM2_RP2350_all_v1.zip](https://github.com/user-attachments/files/18051590/MURM2_RP2350_all_v1.zip)

javavi, 2024-12-08:
I can say right away:
- PWM sound does not work on the VGA firmware for RP2350, but it works on the firmware for RP2040.

I continue testing, I will write comments here.

javavi, 2024-12-08:
PWM sound does not work on the VGA firmware for RP2350, but it works on the VGA firmware for RP2040.
PWM sound does not work on the HDMI firmware for RP2350, but it works on the HDMI firmware for RP2040.
Otherwise, everything seems to be working fine.

javavi, 2024-12-08:
> MURM2_RP2040_all_v1.zip
> MURM2_RP2350_all_v1.zip

PWM sound does not work on the VGA firmware for RP2350, but it works on the VGA firmware for RP2040.
PWM sound does not work on the HDMI firmware for RP2350, but it works on the HDMI firmware for RP2040.
Otherwise, everything seems to be working fine.

javavi, 2024-12-08:
[image]

javavi, 2024-12-08:
[image]

javavi, 2024-12-08:
`set(ZX-MURMULATOR2_VGA_PWM_Audio_defines
  # Speaker pin audio
  SPK_PIN=9
  # Buzzer audio out
  # BZR_PIN=9
  # STEREO AY audio
  AY8912_ABC_STEREO
  # Left chanel audio out
  AY8912_A_PIN=11
  # Right chanel audio out
  AY8912_C_PIN=10
)`

fruit-bat, 2024-12-08:
I'm having trouble with my audio fifo. Not sure why... cross core memory access/caching/me being stupid...

fruit-bat, 2024-12-08:
The samples go in, but the ISR does not see the indexes as updated. I've tried marking the indexes as volatile and a few other tricks but nothing working  so far.

javavi, 2024-12-08:
So the problem is only in the absence of PWM sound for both chips, both RP2040 and RP2350
I tried replacing port definitions in the main code and compiled firmware for RP2040 and I had PWM sound.

javavi, 2024-12-08:
> I take it you are not getting pwm sound on any rp2350 build?

Yes. That's right, there was no sound only on firmware for the RP2350 chip

javavi, 2024-12-14:
> The samples go in, but the ISR does not see the indexes as updated. I've tried marking the indexes as volatile and a few other tricks but nothing working so far.

Well, right now I can only help with testing the firmware.

fruit-bat, 2024-12-16:
Here are some RP2350 builds. The PWM audio is mono for now on GPIO 10. 

[MURM2_RP2350_all_v2.zip](https://github.com/user-attachments/files/18154350/MURM2_RP2350_all_v2.zip)

javavi, 2024-12-16:
Yes, now the PWM sound on the RP2350 is in one channel.
As I understand, you make the PWM sound using PIO, can you explain the reason for this approach ?

fruit-bat, 2024-12-16:
My use of a timer interrupt for the audio was unstable at high frequencies, and on the RP2350 memory access from the callback does not seem consistent with the main thread. Using the PIO has some big advantages as the PWM frame and sample generation are now in sync and buffered by the PIO FIFO (I think it sounds pretty good). Also the PIO block is clocked to regulate the emulation speed. The drawback of this approach is that implementing stereo is more complicated and I have not attempted it yet.

fruit-bat, 2024-12-16:
Is gpio 9 mixed onto both L and R? Should I move the mono ouput to gpio 9?

javavi, 2024-12-16:
> Is gpio 9 mixed onto both L and R? Should I move the mono ouput to gpio 9?

Yes, if the sound is mono with the addition of a beeper sound, then it can be output to the GP9.
But what if it won't be stereo?

fruit-bat, 2024-12-17:
Well, for now I2S will be stereo, PWM will be mono. I've got a bunch of tidying up to do before this branch can get merged back.

See if these sound ok?

[MURM2_RP2350_PWM_v1.zip](https://github.com/user-attachments/files/18171646/MURM2_RP2350_PWM_v1.zip)

The RP3250 has an extra PIO block so I can have a go at stereo using 2 state machines... supporting all the different build variants is now more difficult than the actual code. Super busy with 'real' work at the moment so may be in the new year.

javavi, 2024-12-17:
> Well, for now I2S will be stereo, PWM will be mono. I've got a bunch of tidying up to do before this branch can get merged back. 
> See if these sound ok?

Yes, everything works like that with PWM mono sound.
They write that in SDK 2.1 for RP2350 there are some problems with interrupt timers and this should be fixed, maybe then it will be possible to make a regular PWM sound?

About the menu. 
Can you specify the build configuration in the VER: line ? 
Voltages Vcor:x.xV ​​and Vsys:x.xV in the CPU: line?

closed 2024-12-27 by fruit-bat