fruit-bat/pico-zxspectrum #190: folder scan looping?

8tpercent, 2024-11-24:
Using the latest update on a waveshare with 32gb microsd.

I added some more snaps/tapes to the folders

snapshots - 748 items
tapes     - 1296 items

hit 5 to refresh.  It says scanning on-screen... after about 20 seconds, a few filenames in a list briefly show up on screen and then vanish.

It's as though it's looping.

I tried the same refresh on both snaps folder and tapes and it's doing the same thing.

I can't go back... or return back to the main menu.

If I power off, when I reload.... the folders shows an empty list, or if I time the power off right, I get a list files.

Not sure how I can debug.

fruit-bat, 2024-11-24:
I think you have a much bigger collection of tapes/snaps than me, so that *might* be the problem.... maybe try reducing the numbers a bit and see it makes any difference (I've just tried a version that is similar to the waveshare build and ii works find on the latest build). I'll have a look at the code and see if there is anything obviously stupid I have done!

javavi, 2024-11-24:
There are also problems if there are many deleted files in the folders. It is better to first format the SD card and then write a small number of necessary game image files.

8tpercent, 2024-11-24:
bit more info...

132 files fails to scan using key 5 to rescan
92 files fails to scan using key 5 to rescan
83 files ok to scan using key 5 to rescan

724 files in one snapshot folder fails to scan using  key 5 to rescan

I went into the folder, and deleted:

.dcache
.dcache.tmp

Rebooted the system.

724 Files scan fine on a startup scan (before spectrum screen has loader)

go to menu, hit 5 to re-scan and it doesn't scan this time.

Odd.

fruit-bat, 2024-11-26:
This is mostly a note so I remember what I have tried....

Just 'refreshed' a folder with 2360 tape files, which worked but took a long time, around 4 minutes.

[To keep memory usage down I have to keep writing blocks out to the SD card temp file, hence the slow sort time.
Perhaps something else is going wrong, like really long file names.]

retrofan11, 2024-11-27:
Yes, I always use shorter file names, so far I have had no problems with reading files and subfolders on my sd card, except for a longer wait for a large amount of files, but only during the first scan.

8tpercent, 2024-11-28:
Hi. Same result. I have wiped the microsd and recorded the folders wuth the same result.

A full boot scan works... 5 refresh just loops.

I've put my test folders zipped here if you want to test...let me know if you don't need them or have downloaded them and then I'll remove the file....

https://icedrive.net/s/1tXgfYCTauzV42hy4Sv3NRbiy2GS

fruit-bat, 2024-11-30:
Supplied folders break my build too. Having either the snapshots or tapes folder populated works so probably a lack of memory /memory cleanup issue.

fruit-bat, 2024-12-01:
Any luck with this one...

[ZxSpectrumPiZero_720x576x50Hz.zip](https://github.com/user-attachments/files/17968499/ZxSpectrumPiZero_720x576x50Hz.zip)

It's a tiny change that clears down .dcache.tmp before sorting... if one happens to be lying around. It's now working fine on my breadboard build... maybe there are other considerations with the waveshare board.