fruit-bat/pico-zxspectrum #194: Overclock problem

bobricius, 2024-12-05:
I still have the problem that about 10% of my Pico boards can't work at 250Mhz,
I have a simple program that flashes an LED, which is not capable of working at such a frequency, I don't use it to be sure.
I consulted with Mr. Memecek
He suggests to slow down the FLASH clkdiv = 6

If that didn't help, he also suggested increasing the voltage VREG_VOLTAGE_1_30

// set flash speed
void __not_in_flash_func(SSI_SetFlashClkDiv)(int clkdiv)
{
	u32 i = save_and_disable_interrupts();
	ssi_hw_t* ssi = ssi_hw;
	ssi->ssienr = 0;
	ssi->baudr = clkdiv;
	ssi->ssienr = B0;
	restore_interrupts(i);
}

fruit-bat, 2024-12-07:
I have never had a board fail with the overclock, which I guess make me lucky! 

Couldn't get the above to compile, but according to the docs, I can add:

add_compile_definitions(PICO_FLASH_SPI_CLKDIV=6)

to the make file...

It's hard to know if it is having any effect. Do you have a particular firmware I can build for you to test?

javavi, 2024-12-07:
Basically, the problem with overclocking Pico is not in the RP2040 chip itself, but in the flash memory chip, which can be of different quality, especially on Chinese Pico clones.
And yes, when overclocking the RP2040 chip, to ensure stable operation, you have to lower the flash memory frequency with a divider.

fruit-bat, 2024-12-07:
> Basically, the problem with overclocking Pico is not in the RP2040 chip itself, but in the flash memory chip, which can be of different quality, especially on Chinese Pico clones. And yes, when [...]

Yes, I am sure you are correct. I wondering if the PICO_FLASH_SPI_CLKDIV=6 compile time define actually changed the divider (not whether it was needed). If you have a board you know fails currently I could make a build with the new setting to see if it helps.

fruit-bat, 2024-12-07:
(I still test builds against the boards your sent me... and they always work... which makes it hard to tell if I am solving a problem or not with the above change).

fruit-bat, 2024-12-08:
(Note for me)

See for modifications need to the RP2350 
https://github.com/raspberrypi/pico-sdk/issues/1903

bobricius, 2024-12-08:
Mr. Nemecek did not know how to set the divider in the original SDK, so he also sent this piece of code. I am using the original PICO but I have one board from Waveshare (I have used them for years and they worked, but this one that I have already soldered is faulty) which does not start reliably. It is a PICOZX LCD, VGA version with a 7x7 matrix. When the board already works at such a frequency, then it works all the time. The boards I sent you were tested at 270MHz. I still have the last few pieces of PICO1, and when they run out, I will start using PICO 2. In any case, it would be worth testing whether this divider helps with reliability. Thank you.

fruit-bat, 2024-12-15:
@bobricius see how you get on with these. In theory they have a divider of 6:

[ZxSpectrumPicomputerZxLcd.zip](https://github.com/user-attachments/files/18139668/ZxSpectrumPicomputerZxLcd.zip)

I think the underlying problem may be a little more involved than just the divider. My use of a timer interrupt for the audio seems unstable at high frequencies, and on the RP2350 memory access from the callback does not seem to work at all! I have re-written the PWM code for a mono output to use a PIO block. This has some big advantages as the PWM frame and sample generation are now in sync and the sound is really good from the little speaker. Also the PIO block is clocked to regulate the emulation speed. The drawback of this approach is that implementing stereo is more complicated and I have not attempted it yet... but I don't think this affects any of your boards.

Even if this does not help on the slow board please try it on one that does work and let me know what you think of the sound, I think you will like it :-)

bobricius, 2024-12-15:
Thanks, Tested on this bugy board but it not help, not start reliable, every 3-4 resets not boot.
I try replace flash chip and it is worse .... boot after 3-4 resets same result also on year old 200mhz firmware.
Probably bad board.
Sound looks perfect.

fruit-bat, 2024-12-17:
Just in case it helps here is a version with a divider of 8 (assuming the build is taking any notice of my setting) and core voltage of 1.3v

[ZxSpectrumPicomputerZxLcd.zip](https://github.com/user-attachments/files/18171512/ZxSpectrumPicomputerZxLcd.zip)

This version has a report of the VSYS voltage on the menu. Not quite battery % but its a start.

Phil

bobricius, 2024-12-29:
It not help, still is more freezed like started, last attempt if you have some time. Try overvoltage, set divider and frequency around 200mhz, no care on VGA I have LCD only on board.

fruit-bat, 2024-12-31:
Here we go...

[ZxSpectrumPicomputerZxLcd_200Mhz.zip](https://github.com/user-attachments/files/18281065/ZxSpectrumPicomputerZxLcd_200Mhz.zip)

In theory, all builds now have FLASH clkdiv = 6, but I have not checked this with a scope.

... and so I remember what I changed...

<pre>
diff --git a/src/ZxScanlineVgaRenderLoop.cpp b/src/ZxScanlineVgaRenderLoop.cpp
index 636c757..3213514 100644
--- a/src/ZxScanlineVgaRenderLoop.cpp
+++ b/src/ZxScanlineVgaRenderLoop.cpp
@@ -14,7 +14,7 @@
 
 const scanvideo_timing_t vga_timing_640x480_60 =
 {
-    .clock_freq = 25000000,
+    .clock_freq = 20000000,
     .h_active = 640,
     .v_active = 480,
 
diff --git a/src/picomputer/picomputer_st7789/CMakeLists.txt b/src/picomputer/picomputer_st7789/CMakeLists.txt
index 81d8dd5..01dc9ae 100644
--- a/src/picomputer/picomputer_st7789/CMakeLists.txt
+++ b/src/picomputer/picomputer_st7789/CMakeLists.txt
@@ -123,7 +123,9 @@ set(zxspectrum_zx_lcd_640x480x60Hz_defines
   # TODO work what this is
   PICO_SCANVIDEO_MAX_SCANLINE_BUFFER_WORDS=700
   # Speaker pin audio (@250MHz 250000000 / (1000 * 3 * 2))
-  PICO_PWM_AUDIO_FREQ=41666
+  # PICO_PWM_AUDIO_FREQ=41666
+  # Speaker pin audio (@200MHz 200000000 / (1000 * 3 * 2))
+  PICO_PWM_AUDIO_FREQ=33333  
 )
 
 set(zxspectrum_zx_lcd_720x576x50Hz_defines
diff --git a/src/picomputer/picomputer_st7789/main.cpp b/src/picomputer/picomputer_st7789/main.cpp
index 2d07289..33ef728 100644
--- a/src/picomputer/picomputer_st7789/main.cpp
+++ b/src/picomputer/picomputer_st7789/main.cpp
@@ -48,7 +48,7 @@
 
 #define LED_PIN 25
 
-#define VREG_VSEL VREG_VOLTAGE_1_20
+#define VREG_VSEL VREG_VOLTAGE_1_30
 
 struct semaphore dvi_start_sem;
</pre>

bobricius, 2025-01-04:
Thanks, still the same behavior, it probably won't be a problem with the frequency, but I hit it on a really very bad chip. thanks for the effort, I'll try to desolder it, and then smash it with a hammer