fruit-bat/pico-zxspectrum #197: Quick save key combo on picozx board

shiprat-code, 2024-12-12:
Hi 

I’m using a picozx board inside a spectrum+ case . Is there a key combo to take a snapshot save? TIA

fruit-bat, 2024-12-15:
No, I don't think so... adding key combinations on the spectrum keys seems to end up causing trouble for someone... how about I add an entry to the top level menu?... like press menu then Q

shiprat-code, 2024-12-15:
That would be perfect, thank you so much .

fruit-bat, 2024-12-17:
Code is written but I have a bunch of tidying up to do before I can do a build... maybe a few days.

shiprat-code, 2024-12-17:
Thank you 

very much appreciated. I’m in no rush at all , happy to wait . Thanks again

fruit-bat, 2024-12-27:
New option should be available in the latest builds

shiprat-code, 2024-12-27:
Thanks so much . Works perfectly . Now I can use my 48k plus keyboard to write programs and save them easily !