fruit-bat/pico-zxspectrum #198: Games feel fast on a Bobricius board

Loccy, 2024-12-19:
Have been looking through past issues and docs trying to see if I should expect more accurate emulation than I get, but it's not entirely clear. Issue 113 seems to suggest that while it's not cycle accurate it should be more accurate than my experience has been. I've got a Bobricius ZX48/128 board in an old rubber key case. Am running out of VGA, using the 60Hz version as I don't like the way the 50Hz looks on most of my screens (big black borders everywhere). I would say that everything is about 10-20% faster than it should be, tested on Manic Miner and Cybernoid. AY sound has some curious distortion too on the Cybernoid music - hard to describe but chunks of the percussion is well off.

I've also tried the 50Hz build and that seems the same, with the added "bonus" of those big black borders.

Is this a normal experience or is there something I'm missing? Was hoping this would be a direct swap-in for an original Spectrum board, at least good enough to pass muster for most games, but this is a bit of a show-stopper :(

fruit-bat, 2024-12-20:
Please understand that this is not a commercial project.  I wrote a bit of software, which I thought people might enjoy.  Where people have asked me to add support for their boards, I have had a go. 

The RP2040 is a microcontroller,  and there are limitations in what can be achieved.  Possibly, other projects have done better.  

There is no frame buffer, so on the 60Hz builds, interrupts occur at this rate.  There is a menu option to count cpu cycles... but either way, there will be inaccuracies. The 50Hz builds are a little better... but 50hz on a modern display is not the same as an old CRT.

Loccy, 2024-12-20:
Oh, no, don't get me wrong, that wasn't intended as critical - I was just trying to find out whether what I'm seeing is normal or not. I've got the CPU cycles option enabled so I was hoping to see something that felt close to the original. Am not super bothered about cycle level accuracy. I just wanted to know whether it is "normal" for (say) Manic Miner to feel 10-15% faster than stock? And also is it normal for this to be the case seemingly regardless of whether I use 50Hz builds or enable the CPU cycles counting?

fruit-bat, 2024-12-20:
Possibly a couple of issues. The emulation is clocked against the timer interrupt for the audio which I have been having a few other issues with.  I am working on some pio versions without an interrupt which may work better.

I'm also a bit lax in counting memory access wait states accurately, which may correspondence to a speedup.

Loccy, 2024-12-20:
So, see below my incredibly unscientific testing using the Manic Miner air supply as a timer.

[image]
[image]

Left hand side Spectrum and screen is original hardware, right hand side is a Bobricius board running the two different from commit abfd402. Top pic is at 60Hz, bottom at 50Hz, both on a Bobricius board running the ZxSpectrumPicomputerVgaAukBob_640x480x60Hz.uf2 firmware, apparently from commit abfd402. Both running at 3.5MHz. On the 60Hz I had "count CPU cycles" on, on the 50Hz I had "sync to display".

50Hz or 60Hz, they both seem to run at a comparable speed regardless insofar as when the air runs out on the Pi-based Spectrum the real Spectrum's air has just got into the red. Obviously I started the games both at the same time. So the speed is consistent (to the end-user's eye, at least) across the different refresh rates and even across the two different CPU interrupt modes. That consistent speed is just faster than it should be.

fruit-bat, 2024-12-22:
Thanks for the detail. Please would you test these builds for me and let me know if they match the real hardware better (I've not owned a real Speccy for a long time)?

This is a just a hack, I've slowed it down overall by about 10% (or tried to).

[ZxSpectrumPicomputerVgaAukBob_90percent.zip](https://github.com/user-attachments/files/18222862/ZxSpectrumPicomputerVgaAukBob_90percent.zip)

Loccy, 2024-12-23:
Thanks for that - just to let you know it won't be until the New Year that I'll be able to test this, as my Spectrums are at the office and I'm away for the hols. But will check it out as soon as I get back.

fruit-bat, 2024-12-27:
Note to myself: I need a framework in which I can easily add wait states to memory accesses, based on memory address.  Probably needs changes to the Z80 emulator.

fruit-bat, 2024-12-28:
Note to myself: https://en.wikipedia.org/wiki/Contended_memory

Loccy, 2025-01-02:
So, we still seem to be running a touch fast using the same highly scientific testing methodology:

[image]
[image]

There didn't seem to be any difference (timing-wise) between 50 and 60Hz. When the emulated Spectrum ran out of air, the real Spectrum's air was just a pixel or so to the right of the S in "High Score" in both cases.

It is much closer - possibly getting close enough that I wouldn't have noticed if this had been the first build I'd ever run (it probably wouldn't have occurred to me to put it side-by-side with a real Spectrum). But still a tad off. 11%, perhaps?

Loccy, 2025-01-08:
> Note to myself: https://en.wikipedia.org/wiki/Contended_memory

This one gives a LOT more detail (most of which I don't understand in the slightest) https://sinclair.wiki.zxnet.co.uk/wiki/Contended_memory

fruit-bat, 2025-01-25:
So, adding in extra wait states turns out to be trivial...

```
   inline uint8_t readByte(uint16_t address) {
+    _Z80.cycles+=4;
     return *memaddr(address);
   }
```
Now, need to apply the correct number at the correct time.