fruit-bat/pico-zxspectrum #199: Any Plans for the +2 and +3 ?

RattyDAVE, 2024-12-20:
Is there any plans to impliment the +2 and +3 ?

That would be really good.

Thanks

retrofan11, 2024-12-22:
In my opinion, its necessary to set realistic goals and keep in mind that this emulator works on a Pico microcontroller, not on a microcomputers like Pi Zero, 2, 3 etc.
Second, we already have the original Siclair 128k model supported, which is the most compatible with all software, than the later Amstrad models.
I believe that its not too much of a problem to add +2 rom instead of 128k, but I don't see the point.
Its much more important that the focus be on the accuracy of classic 48k/128k models (or at least 48k), above all timig issues, which is why a significant number of quality games do not work properly.
We already talked about this in other issue topic and if its possible to make a progress in this direction, we will have a top-notch emulator, although it is already very good.
Cheers.

fruit-bat, 2025-01-11:
Apologies, no plans at the moment... but it is still a nice idea. Getting more accuracy for the existing models is already tricky, so even if the roms were added I'm not sure faithfull the emulation would be. 

If I were aiming for more accuracy,  I might need quite a different approach... like maybe use 2 chips. One for the emulator and another to adapt to various displays and audio solutions.  At this point it might just make more sense to use a more capable soc, like a pi 0.

RattyDAVE, 2025-01-11:
Not a problem. I have a few RPI sitting around.