fruit-bat/pico-zxspectrum #208: VGA Breadboard shows 2 different sets of pinouts

adsorptionenthalpy, 2025-04-02:
Which circuit diagram is correct?

[docs/ZxSpectrum4PinAudioVga1111Ps2.md](https://github.com/fruit-bat/pico-zxspectrum/blob/main/docs/ZxSpectrum4PinAudioVga1111Ps2.md)

fruit-bat, 2025-04-02:
Ah yes, apologies, the video pins start at GP12 and end with CS on GP16.

I need to update the diagram. I should also check the colour components are written in the correct order.

fruit-bat, 2025-04-02:
See https://github.com/fruit-bat/pico-zxspectrum/blob/main/src/vga/CMakeLists.txt

Note that RGBY is not really supported by the scanvideo library so don't trust the RGB order.

fruit-bat, 2025-04-06:
Hopefully correct pins now. I also checked the RGB order and it looks correct too.

adsorptionenthalpy, 2025-04-06:
> Hopefully correct pins now. I also checked the RGB order and it looks correct too.

Thank you for the quick turnaround. This is much clearer. Cheers!

closed 2025-04-19 by fruit-bat