fruit-bat/pico-zxspectrum #209: Spotpear HDMI Board

Beathoven666, 2025-04-13:
Hello,

Many thanks for this amazing project. Would it be possible to create a build for this Spotpear HDMI Board? Unfortunately, as I have checked, their DVI pinout doesn't seem to match any of the existing builds...

https://spotpear.com/index/study/detail/id/971.html

fruit-bat, 2025-04-14:
If you don't mind testing the builds, I could probably come up with something for you; It can take a few tries to get things working.

You probably want an SD card wired up as well; do you have any preference as to where that would connect?

Beathoven666, 2025-04-14:
I will be happy to test it. I have an Aliexpress SD card adapter that has this sequence of signals: CS, SCK, MOSI, MISO. But since I will have to connect it to PICO with wires anyway, the order doesn't really matter. You can even leave it as it is on the breadboard build on GP2-5.

fruit-bat, 2025-04-18:
GP2-5 for SD card. Pin 7 for PWM audio. 

Let me know if anything works.

[spotpear_test_v1.zip](https://github.com/user-attachments/files/19811550/spotpear_test_v1.zip)

fruit-bat, 2025-04-18:
https://github.com/fruit-bat/pico-zxspectrum/pull/210

fruit-bat, 2025-04-18:
> Thank you! It seems to work. I just can't get SD to work. Are the signals
> on the GP2-5 in the original breadboard build order or as I listed them
> above?

Ah, should have said...
   SDCARD_PIN_SPI0_CS=5
   SDCARD_PIN_SPI0_SCK=2
   SDCARD_PIN_SPI0_MOSI=3
   SDCARD_PIN_SPI0_MISO=4

Beathoven666, 2025-04-18:
That's how I have it. Probably something to do with my dirty quick construction :-) I will have to investigate.

fruit-bat, 2025-04-18:
On some of the builds I use a PIO block for the SD card... there is a small chance this will work...

[spotpear_test_v2.zip](https://github.com/user-attachments/files/19814071/spotpear_test_v2.zip)

Beathoven666, 2025-04-18:
Thanks. No luck. My wiring looks alright. It could be something with SD card itself, I guess. I only have a couple of large ones, although I have tried only having a 2GB partition on them.

fruit-bat, 2025-06-03:
Just on the off chance this has started working...

[spotpear_test_v3.zip](https://github.com/user-attachments/files/20580896/spotpear_test_v3.zip)

Beathoven666, 2025-06-06:
Thank you. Still no luck. I guess I will have to put it on hold until I get ahold of a lower-capacity SD card.