fruit-bat/pico-zxspectrum #212: Keyboard

kestasa, 2025-05-16:
Hello, have an issue using keyboard, SHIFT works only for letters but did not work for numbers or other symbols like    ' " ' or ' : ' and so on... Tried SHIFT, SHIFT+CTRL, CTRL, ALT but did not work :(

kestasa, 2025-05-17:
Look like it was a USB keyboard issue, found one which works great. Also my RP2040 with emulator start only after RUN button is pushed. Also have few more questions:
1. It is possible to implement an option to test entire keyboard (to see what keys what function do)?
2. As I can see emulator also tries to play RZX format but file can't be loaded because emulator waits for loader first and RZX do not have one.
3.  Is it possible to add option to immediately load tape formats (TAP, TZX)?
4. Is it possible to add support for SNA, RZX, TRD, SCL formats?
5. Do you have future plans to implement [rp2040-psram](https://github.com/polpo/rp2040-psram) function for this project?

Zorglupy, 2025-05-18:
Hello, Strangely, I have the same problem as the one described by Kestasa. But I am using a Spectrum matrix keyboard (not USB). The keys that work in combination with the SHIFT key are those found in the same row (ROW6) and column (COL3). The same behavior occurs for combinations with SYM. The only ones that work are those in ROW5 and COL1. I use th PICO ZX48/28 firmware.

fruit-bat, 2025-06-03:
Don't really have any plans, every now and then if I get time I add some things in.

A keyboard test is a really good idea... I'm not quite sure how it would look as the spectrum and the menus sort of work differently... but I will have a think... it's something that would have been useful for me about 1000 times while writing the software.

Is RZX playing back key sequences?

It might be possible to create a load immediately for TAP files... but it might slow the emulator down... I think the trap for this needs to live in the memory read which gets called lots. 

PSRAM is a blessing and a curse... you get lots of memory but page operations are really slow. I have to keep as much of the emulation in on chip ram as possible or it interferes with the video and audio.... you can literally hear it.

kestasa, 2025-06-04:
Noticed one more issue (maybe it is HW related, maybe not): if I want to connect keyboard and gamepad at once I need connect USB hub but then after turning power on I must push RUN button otherwise it will not start. After it started everything works even if will reset from emulator menu. If I unplug USB hub board starts immediately, no need to push RUN button. If connect newer hub or native USB C - it will not start at all, you need to use adapter for older hub or replace cable to avoid adapter...

kestasa, 2025-06-10:
Question about joysticks/gamepads: if I connect joystick/gamepad (windows recognized as generic usb gamepad) it works but if connect device (recognized by windows as PS3/PC gamepad) it did not working on Pi zsero... Any chances to fix this?

fruit-bat, 2025-06-15:
The emulator *mostly* works with joysticks that generate HID report descriptors, which announce the capabilities of the device. Some manufactures decided that it would be better to hide how their controllers work, so without a custom driver they  won't function; hence why the PS3 controller doesn't function. It's possible, but non-trivial to extend the code in here https://github.com/fruit-bat/pico-hid-host to add custom drivers. An Xbox controller driver was kindly donated. I might have a go at a PS3 controller sometime, but struggling for time so probably not soon.