fruit-bat/pico-zxspectrum #216: no display out on hdmi on never tc

pauldude12, 2025-07-18:
is always
best to connect in the order 1-2-3
Some televisions put out a low current 5v power over the HDMI cable-ifyou plug the HDM I in before the USB power -it will confuse the device and it won't boot. If that happens take the HDMI cable out, then the power cable and wait 5 seconds. 

The PC needs a resister on the 5 volt hdmi output pin of thd pico to resolve this

fruit-bat, 2025-07-20:
Thanks for sharing this.

pauldude12, 2025-07-20:
Your welcome

pauldude12, 2025-07-20:
Ps woks on a sharp 50 inch tv no issues
TCL 50 inch no signal
LG standred 32 inch work ok

pauldude12, 2026-03-14:
The display only work when you use the gio pins as the usb port is only to flash the firmware usually like to xstation or playstation peko where you flash firmware but use the gio pin to power the piko via the playstation power pack it neeed a power jack and a gnd and wire wired to positive rail