fruit-bat/pico-zxspectrum #220: floppy disks

diymarcus, 2025-09-20:
Hello Dear fruit-bat
I'm delighted  with your software. It takes me back to my younger days when I had a ZX Spectrum. I do have a question:  I still have many floppy disks and cassettes, some containing my old projects—nothing serious, but I’d love to be able to read them again. Is there any possibility of adding support for a physical 3.5-inch floppy drive and audio input in your project? Thanks in advance, and best regards.

fruit-bat, 2025-09-27:
Audio input is already available on some of the builds. I've not played around with it for a while now, so there may be the odd tweak needed. It needs a signal conditioning circuit before connecting to the appropriate gpio pin. 

 I don't know much about the disks that the later versions of the spectrum adopted. It might be possible, but it might also be quite difficult.  What model spectrum / drives did you have?

diymarcus, 2025-09-27:
Hi, thanks for your reply. I have a couple of TR-DOS 3.5" disks. I’ve already used the Greaseweazle floppy adapter on my PC to create .TRD images, which I can read in the emulator. However, I’d really like to be able to read the disks directly in the emulator. I think this would require a driver to connect the Greaseweazle floppy adapter directly to the emulator. in this chase the emulator will be axact like a real ZX Spectrum. In this case, the emulator would behave exactly like a real ZX Spectrum. I read that the Amiga emulator already has support for Greaseweazle, which can read and write the raw flux from a floppy drive. Thanks in advance